dgtaheno/artillery-Genius-Genius_Pro-X1-Marlinfw #4: Unload button error

alfem, 2022-09-18:
After flashing your marlin firmware and updating TFT, if I click "unload" a message pops up: "unknow command: "M702 T0"

dgtaheno, 2022-09-18:
Yes, just experienced it a few days ago, I'll check it!

vertice3d, 2023-02-01:
Could you find a solution to this issue?

dgtaheno, 2023-02-01:
Sorry for the long time without response on this issue @alfem and @vertice3d , I have been a bit busy lately.
I have investigated about the issue and I´m quite sure that it is a problem with dependencies in Marlin FW as mentioned here:

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#marlin-dependencies

Despite I found the root cause, I need to find some time(energy) to correct the code, test it and update it.

Please be a little bit more patient.

Best regards.

dgtaheno, 2023-02-01:
Hello @alfem & @vertice3d ,

I felt a bit ashamed and corrected the code now :-).

I just tested and now unload and load functions seem to work fine.

Good think is that with new TFT firmware, if already installed, you don´t need to open the printer anymore to upload the firmware.

Just go to  Menu->Settings->Connection->ON

upload the firmware via Cura(plenty manuals online and quite intuitive).

And you should be able to have it running quite fast.

Please keep reporting bugs and I´ll try to be faster.

Best regards.

closed 2023-02-01 by dgtaheno

vertice3d, 2023-02-01:
Thank you very much for your help, I'll try it