dgtaheno/artillery-Genius-Genius_Pro-X1-Marlinfw #7: why so slow?

jamarju, 2022-12-19:
Hello and thanks for your excellent work! I was wondering the following:

```diff
< #define DEFAULT_MAX_FEEDRATE          { 300, 300, 50, 40 }
---
> #define DEFAULT_MAX_FEEDRATE          { 200, 200, 20, 40 }                     // DIGA-Tech: decreased values
```

Why decreased values? Especially the Z axis runs overly slow IMHO.

dgtaheno, 2023-01-10:
Hello @jamarju ,

first of all sorry for the late reply, I was on vacation :-).

Glad that you thank you for taking the time to go through the code and revise it.

For being honest I did not worked a lot in the configurations and in some cases just took the ones in Marlin configurations example for Artillery Genius V1 (see attached)

If I´m not wrong the original author for theses configurations is @thisiskeithb, maybe he can give you better answers than me :-).

Checking over there seems that the speeds that you mentioned should also work.

If you need to discuss further feel free to reach out.

Best regards.

thisiskeithb, 2023-01-10:
It’s been a couple years, but these settings were ported from Artillery’s original firmware, wherever that may have come from at the time.

Feel free to update settings in your fork as you wish.

dgtaheno, 2023-01-10:
Thank you for your valuable feedback @thisiskeithb!!

closed 2023-01-10 by dgtaheno

dgtaheno, 2023-02-01:
In updated version I can confirm in configurations.h the speeds are:

#define DEFAULT_MAX_FEEDRATE          { 300, 300, 50, 40 }

Hence it should be correct now.