dgtaheno/artillery-Genius-Genius_Pro-X1-Marlinfw #10: M900 not working

GeestSoft, 2023-04-16:
M900 command is not working for "Linear Advance 1.5" readout and setting. Why ?

dgtaheno, 2023-04-17:
Hello @GeestSoft ,

You are right, Linear advance is not enabled in Marlin2.1.2.hex, so I have compiled another firmware version([Marlin2.1.2-la.hex](https://github.com/dgtaheno/artillery-Genius-Marlinfw/blob/78241bf213f351f7d83aada767a419af6181e339/Marlin%202.1.2-la.hex)), which has this functionality enabled, the counter side is that S_CURVE should be disabled.

Please note that I have not tested this version, if you find any issue, please let me know so I can correct it.

If everything works as expected, the feedback would be also be appreciated.

BR

closed 2023-04-17 by dgtaheno

GeestSoft, 2023-04-17:
Hi dgtaheno,
Thanks! for the fast response. I will test the update and let you know how it works.

dgtaheno, 2023-05-25:
As an additional clarification.

We have 2 different firmwares:

**Marlin2.1.2-la.hex **
which has enabled linear advance and disabled s_curve(they are not compatible)

**Marlin2.1.2.hex **
which has disabled linear advance and enabled s_curve

So you can choose between both of them, so you can use as per your convenience.