dgtaheno/artillery-Genius-Genius_Pro-X1-Marlinfw #11: buffer size

fernandex00, 2023-06-10:
im having issues with octoprint and the solution is changing the buffer size in marlin configuration, this two values
BLOCK_BUFFER_SIZE from 16 to 32
TX_BUFFER_SIZE from 0 to 32

do you think is posible?

dgtaheno, 2023-08-18:
Hello @fernandex00 ,

Thank you for your feedback and my apologies for coming back to you that late, I have been quite busy lately.

I made the modification you suggested in the firmware and the new test firmware is:

https://github.com/dgtaheno/artillery-Genius-Marlinfw/blob/Marlin-2.1.2/Marlin%202.1.2_BufferFix.hex

I´d appreciate if you could test it and if it works I will apply this change in the main firmware permanently.

Best regards.

closed 2023-08-18 by dgtaheno

fernandex00, 2023-08-19:
thank you, i will test it!

fernandex00, 2023-08-20:
hi im having trouble updating the firmware, i dont know if this is causing by the firmware itself, already turn on the conection and chek my usb cable but i get thi error. "could not open port and can't open device "\\.\COM5

dgtaheno, 2023-11-06:
> hi im having trouble updating the firmware, i dont know if this is causing by the firmware itself, already turn on the conection and chek my usb cable but i get thi error. "could not open port and can't open device "\\.\COM5

Hello,
I just experienced the same problem once I have the new TFT firmware, the problem is that Marlin baud rate is set now at 250000, but prisa slicer and others are configured to 115200 for the communication.

Solution:
Before connecting the printer via usb to the computer it is required to:
Menu->Settings->connection->S.ports there change 1.Printer parameter from 250000 to 115200 and go back to Connection menu and click ON and after that disconnect.

There you can upload the new firmware.

After it is done updating the firmware, come back click back on OFF in connection and set the baudrate again to 250000 otherwise the motherboard will not connect to the tft.

And with that you should be done.

If you need further assistance don't hesitate to reach out.

Best regards.

fjherna, 2024-03-07:
Hola a todos y particularmente a dgtaheno,

Disculpen que les aborde así sin haber participado antes. He visto un proyecto que hay en GitHub, en el que estoy muy interesado y creo que esta relacionado con este foro.

Tengo una Artillery Genius (básica y de las primeras pues ya hace unos cuantos años que la tengo; creo recordar que fue en Enero 2021).

Reconozco que la tengo bastante abandonada, y justo ahora me he decidido a retomarla.

Les pongo un poco en antecedentes de cual es mi situación actual. Soy totalmente nuevo con la impresion 3D y apenas he hecho unas pocas piezas propias, además de imprimir algunas de thingiverse.
🙂
Lo mas que he hecho ha sido instalar el Marlin 2.0.5.3 y posteriormente compilar yo mismo el Marlin 2.1.x actual para actualizarlo, pues cambie el extrusor original por uno "Trianglelab MATRIX" (del estilo del HEMERA, pero mas barato).

Hace unos días que me plantee buscar una pantalla complementaria para manejar un poco mejor la impresora y el Firmware si fuera posible. 

En mi búsqueda encontré una información interesante en YouTube (https://www.youtube.com/watch?v=H3mHDbtcaxU). Pensé entonces en adquirir una pantalla FYSETC, pero el problema es que en ese articulo menciona solamente Marlin 1.9.0 y 2.0.5.3 Además es de hace tres años y no tiene actualmente actividad. Pregunte por la posibilidad de usar un Marlin 2.1.x con esas pantallas FYSETC o similares, pero no espero que nadie conteste.

En vista de ello seguí buscando y fue como di ayer con su proyecto en el que se menciona un Firmware 2.1.2.1 y unas pantallas BigtreeTech TFT Touchscreen que me parecieron que podrían tener posibilidades con mi Artillery Genius.
En los enlaces que he visto (https://github.com/dgtaheno/artillery-Genius-Genius_Pro-X1-Marlinfw) el proyecto es bastante reciente.

Bueno no me quiero extender mas (quizas ya es demasiado y lo siento), pues mi intención era únicamente una toma de contacto con Vd. respecto a este proyecto, y saber si sigue en marcha.

Por supuesto que me ofrezco a colaborar en lo que hiciera falta, siempre considerando la modesta medida de mis posibilidades y limitaciones.

Reciban todos un cordial saludo,

fjherna1