dgtaheno/artillery-Genius-Genius_Pro-X1-Marlinfw #12: How to check TFT firmware already flashed

ahmadkarlam, 2023-09-04:
Hi, I bought second hand Artillery Genius and I want flash to newer marlin firmware. Is there any way to know that TFT firmware already flashed or not so I don't have to open back cover of my 3d print ?

dgtaheno, 2023-09-11:
Hello,
welcome onboard to Artillery Genius community.

You can check your TFT Firmware version by:

Menu-->Settings-->Info

[image]
[image]
[image]
[image]

closed 2023-09-11 by dgtaheno

ahmadkarlam, 2023-09-12:
Thanks, I already done update firmware and its amazing. Great work @dgtaheno

dgtaheno, 2023-09-12:
Thank you for your feedback!appreciate that you are enjoying it!