dgtaheno/artillery-Genius-Genius_Pro-X1-Marlinfw #13: Source files

kjkent, 2023-09-29:
Hi there! Thanks for your work on the Artillery Genius with Marlin.

Forgive me if I'm being slow, but I can currently only see the compiled .hex files. Have the source files been removed?

Many thanks.

dgtaheno, 2023-10-01:
Hello @kjkent ,

You are not slow at all.

I´m not including the source files on purpose so it is not confusing the people trying to install it.

If you need them for some adaptations just let me know so I can share them with you.

BR

closed 2023-10-01 by dgtaheno

kjkent, 2023-10-01:
@dgtaheno thank you! That makes sense. It would be great to get a copy; I'm using Klipper at the moment but considering changing back to Marlin and I'm keen to check out your optimisations. Perhaps you could upload them to a new branch? Alternatively, my email address is kris@kjkent.dev if you prefer to share over email. Thanks again!

dgtaheno, 2023-10-09:
Hello @kjkent ,

sorry for taking some time answering.
Here you can find the source that I have been working for  the last buffer correction.
If you need clarifications, don´t hesitate to reach out.
[Marlin-2.1.2.zip](https://github.com/dgtaheno/artillery-Genius-Marlinfw/files/12845055/Marlin-2.1.2.zip)

Best regards.
David

kjkent, 2023-10-09:
Thanks David, and no worries at all! I appreciate you sharing your hard work with me.