dgtaheno/artillery-Genius-Genius_Pro-X1-Marlinfw #17: Updating Firmware with OctoPrint?

Austin1, 2023-11-19:
I've been able to stumble my way through one update before and get the TFT and the firmware updated to Marlin using an older/beta firmware. Someone had posted a firmware for the Genius Pro on one of the 3d print sites and I followed along until I got it working. It's been working fine, but I'm a sucker for the latest version.

So I googled and found this repo and was looking to try it out. I was wondering, since I have Octoprint hooked up and working pretty good with SimplyPrint.io, I was wondering if I could push the firmware from it.

However, I have no idea which "method" to use. See screenshot below:
[image]

I tried googling a few of these methods to see which one might be compatible with my Artillery Genius Pro without much luck. I figured I'd ask and see if it's an easy answer. If it works, I'd be happy to write some instructions for your docs on this method!

dgtaheno, 2023-11-26:
Hello @Austin1 ,

Thank you for considering using my fw.

I never used this software but it looks that it should work.

I understand you are using or x2 or genius pro, if that is the case, I'd try with marlinbft as this is the type of file for these 2 printers (.bin)

I'd appreciate if it works fine for you if you could share the steps you followed.

Good luck and looking forward to your feedback!

Best regards

Austin1, 2023-11-26:
Well, I ended up flashing klipper instead. But might go back and will use this if I do. Took me a few days to get a klipper print that worked but just finished my first successful print.

dgtaheno, 2023-11-26:
I heard it works pretty well!hope that you enjoy your prints no matter klipper or Marlin!