dgtaheno/artillery-Genius-Genius_Pro-X1-Marlinfw #18: Will this work with the TFT GD32F305 ?

systemerror-codenotfound, 2023-11-25:
Will this work with the TFT GD32F305 ?

dgtaheno, 2023-11-26:
Hello @systemerror-codenotfound ,

Thank you for raising this topic, as I don't own a genius pro or X2 never though about it.

Googling a bit I think Marlin update will work but I'm afraid this tft is not upgradable yet.

I found this interesting repo where it is explained how to upgrade the firmware.

https://github.com/Trivecta95/Firmware_Flash-Artillery_X2_GeniusPro

Hope that this will answer your question.

Best regards.

systemerror-codenotfound, 2023-11-26:
HI,

Maybe with this?
I know for sure that i can flash that TFT firmware and it works, at least in part, there are some options on screen that crash the printer, but hey, it's something

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2391#issuecomment-1753374468

closed 2023-12-17 by dgtaheno