dgtaheno/artillery-Genius-Genius_Pro-X1-Marlinfw #20: TFT Zip File Invalid?

ultradave9000, 2023-12-15:
I've downloaded the TFT file but I cannot extract it's contents - both 7Zip and Windows report the file as invalid.  Have re-downloaded multiple times.  Has the hosted filed become corrupt?

dgtaheno, 2023-12-17:
It is strange man, just tried and it is working for me
[image]
Maybe it is a problem you have with zip?

closed 2023-12-17 by dgtaheno

ultradave9000, 2023-12-17:
Ah, thanks for checking.  Realised it was how I was downloading it - 'Save Link As' doesn't actually download the zip file (like it did the hex files) - weird.