dgtaheno/artillery-Genius-Genius_Pro-X1-Marlinfw #23: Introduction to the Foro + doubt

fjherna, 2024-03-09:
Hi all and in particularly to dgtaheno,

Sorry for approaching you like this without having participated before. I have seen a project on GitHub, in which I am very interested and I think it is related to this forum.

I have an Artillery Genius (basic and one of the first since I have had it for a few years now; I seem to remember it was in January 2021).

I admit that I have quite abandoned it, and just now I have decided to take it up again.

I'll give you a little background on what my current situation is. I'm totally new to 3D printing and have only made a few pieces of my own, apart from printing some from thingiverse.
:)
The most I have done is install Marlin 2.0.5.3 and then compile the current Marlin 2.1.x myself to update it, so I changed the original extruder for a "Trianglelab MATRIX" one (similar to the HEMERA, but cheaper).

A few days ago I considered looking for a complementary screen to manage the printer and the Firmware a little better if possible.

In my search I found some interesting information on YouTube (https://www.youtube.com/watch?v=H3mHDbtcaxU). I then thought about purchasing a FYSETC screen, but the problem is that in that article it only mentions Marlin 1.9.0 and 2.0.5.3. Furthermore, it is from three years ago and is currently not active. I asked about the possibility of using a Marlin 2.1.x with those FYSETC or similar screens, but I don't expect anyone to answer.

In view of this, I continued searching and yesterday I found your project in which Firmware 2.1.2.1 is mentioned and some BigtreeTech TFT Touchscreen screens that seemed to me could have possibilities with my Artillery Genius.
In the links I have seen (https://github.com/dgtaheno/artillery-Genius-Genius_Pro-X1-Marlinfw) the project is quite recent.

Well, I don't want to go into more detail (perhaps it's already too much and I'm sorry), because my intention was only to make contact with you regarding this project, and to find out if it is still going.

Of course I offer to collaborate in whatever is necessary, always considering the modest measure of my possibilities and limitations.

Everyone receive a cordial greeting,

fjherna1

dgtaheno, 2024-03-09:
Hello @fjherna ,

Welcome and thank you for participating in this repository.

Yes, the project is going but as I´m the only collaborator and being quite busy from the personal/professional side I have not too much time to invest here.

BigtreeTech is the original display that comes with the Artillery Genius(standard version) and X1.

Related to the display you mentioned, I think everything is possible, in FYSETC documentation I found(or at least I think I found) the configurations you should do in Marlin to use this display:

[https://wiki.fysetc.com/Mini12864_Panel/#firmware-config](https://wiki.fysetc.com/Mini12864_Panel/#firmware-config)

In any case I understand that in 3D printing there is people focused in 3D printing and other in modifying the printer, so enjoy the one you like the most and feel free to ask.

Best regards and have a nice weekend.

P.S.: If you want to open this kind of discussions I suggest you to use the discussion part instead of issues, but not a big problem.

closed 2024-03-09 by dgtaheno

fjherna, 2024-03-09:
Hello dgtaheno,

It is really a big surprise to receive Emails from any of the foros related to your project. 🙂

I had not idea of the foros till recently. That is reason why I sent an Email on the 26 Feb 2024 to your Email. I am sorry about it.

Recently I found the foros, and I posted to several of them. I was not sure which one was the correct one.

I really appreciate your replies to some of the questions of my post.

Finally, I decided to order a BTT TFT35-E3 V3.0 Display Touch Screen Two Working Modes from Biqu Equipment. It is on the way to me now.

Thank you for your TIP about using the "discussion part" instead of issues, for this kind of discussions.

Kind regards,

fjherna

https://wiki.fysetc.com/Mini12864_Panel/#firmware-config