dgtaheno/artillery-Genius-Genius_Pro-X1-Marlinfw #32: Filament runout sensor not working Genius PPRO (Marlin 2.1.2.1_PRO.bin)

rcaruso82, 2024-05-06:
Dear all,
Steps to reproduce:
Enabled filament sensor in Features menu and start a print job.
It causes a false positive (the light in the sensor works correctly changin from red to green).
The command M412 is also not recognized by the printer.

https://marlinfw.org/docs/gcode/M412.html

Thank you for your help, best regards!

dgtaheno, 2024-05-11:
Hello @rcaruso82 ,

thank you for trying the firmware.

Today I was investigating about the issue and it seems that the problem might be related to TFT´s FW more than to Marlin´s FW, please have a look on:

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2374

If it solves the problem for you, please let me know, if not please come back here so we can have another look on the issue.

Best regards.

rcaruso82, 2024-05-18:
Hello, thank you for your reply.
The problem is not int the TFT firmware, previously I was using other Marlin version without this problem (with the same TFT).

Is #define FILAMENT_RUNOUT_SENSOR enabled in Configuration.h?

Best regards,

Oneseat1, 2024-05-19:
I am having the same issue filament sensor false runout alert. I've disconnected the sensor wire all the way to the tft, and still getting the alert. Just commenting to track

dgtaheno, 2024-09-29:
Hello Everybody,

First of all apologies for the looooooong response time.

I have prepared a revised FW which I think should work, but as I have no Genius pro(donors would be appreciated for testing... :-)), I could not test the FW, if anyone could try it and share the feedback  and it works I can mark this new revision as the solution for the problem.

Here is the file: https://github.com/dgtaheno/artillery-Genius-Genius_Pro-X1-Marlinfw/blob/2e15512fc10c0bc1f8abb0390b27867747aa1399/Marlin%202.1.2.1_PRO_fil_sens_01.bin

Best regards.

nagyrobi, 2024-11-18:
> Enabled filament sensor in Features menu and start a print job.

Which Features menu? There is no such thing in my Octoprint > Features menu...

dgtaheno, 2025-01-05:
> > Enabled filament sensor in Features menu and start a print job.
> 
> Which Features menu? There is no such thing in my Octoprint > Features menu...

Sorry but cannot get which features menu you are referring to.... I enabled the filament sensor in marlin firmware 2.1.2.1(https://github.com/dgtaheno/artillery-Genius-Genius_Pro-X1-Marlinfw/blob/2e15512fc10c0bc1f8abb0390b27867747aa1399/Marlin%202.1.2.1_PRO_fil_sens_01.bin)  for artillery genius pro.

You can just install it and it should be fixed, but not sure if it is working, as I received no feedback.

BR.

closed 2025-01-05 by dgtaheno