dgtaheno/artillery-Genius-Genius_Pro-X1-Marlinfw #36: 3D TOUCH and bed

vertice3d, 2024-07-03:
Hi there, i already install the version 295c° with BL TOUCH (I USE A 3D TOUCH VERSION), It seems to work fine, but notice that the red light is always flashing, even when printing. The other problem is in the bed, when the bl touch does the mesh, the bed seems to hit when it reaches the top of the front part. When it reaches the back top, does it well without hitting.

Psycomy, 2024-07-13:
Same problem here with m701 and m702 when try to load and unload filament. Artillery X1 with BLTouch.

ProSteelPT, 2024-09-02:
Hi, Marlin 2.1.2.1-X1-BLT.hex
I installed it but I have this problem too!

When I'm printing, BL Touch activates and deactivates, hitting the tip on the piece during printing